openUC2/UC2-GIT #24: How to rigidly mount camera in RaspiCam insert v2?

astraw, 2020-01-13:
Hi, I just printed all the files in `UC2-GIT/CAD/ASSEMBLY_CUBE_RaspiCam_v2/STL` but don't understand how the sensor is supposed to be held at a particular postion without play. Is there some way how the `20_Cube_Insert_RaspiCam.stl` part is held rigidly within the cube that I am missing?

Also, when I look closely at [this picture from the inline holography example](https://raw.githubusercontent.com/bionanoimaging/UC2-GIT/04edb5012f1e433878d4b0439712a9407a652075/CAD/APP_INLINE_HOLOGRAM/IMAGES/Application_Inline_Holographic_Microscopy_v2.png) (zoomed in and annotated below), it looks to me like the RaspiCam insert is rendered a few mm lower than is physically possible.  In particular, if you look at how the gray camera insert overlaps the dark green lid, it looks like the lid is occupying the same volume as the insert, which is physically impossible.

[image]

If I look at [the Camera Cube readme](https://github.com/bionanoimaging/UC2-GIT/blob/504b942e821c90b68e9c870a320eb38d880c2d3c/CAD/ASSEMBLY_CUBE_RaspiCam_v2/Readme.md) for help, this unfortunately does not clarify the situation for me. Also, the last image shows the sensor held by only a single screw, so perhaps a rigid mounting is not very important for this application.

beniroquai, 2020-01-16:
Hey! Thanks for reaching out! The photo you showed is indeed a bit misleading since the rendering indicates an error in the assembly. Once the components are printed it should fit. If you follow [this link](https://github.com/bionanoimaging/UC2-GIT/tree/master/CAD/ASSEMBLY_CUBE_RaspiCam_v2) you'll see how the camera module has to be assembled. The printed insert is "pushed" into the cube (e.g. press-fit or form-fit). You need a set of two screws to mount the Raspi-Camera. 

Is this somehow clear?

astraw, 2020-01-17:
Thanks for the reply.

The components do fit together, that is not the issue. Rather, the issue is that this is "too easy" -- there is substantial play of the insert within the cube. You say these are press-fit or form-fit but at least on my print, there is substantial play. This play is reduced when I insert screws into the cube and the threads of the screws touch the corners of the insert. For what it is worth, with this degree of play, I would not calibrate the camera as I think the camera would shift enough to give substantial error.

beniroquai, 2020-01-17:
Ah. This is interesting. May I ask you which printer you have? It's
supposed to be stiff and not very easy to move.
One thing we were thinking of is having a template to adjust the inner
component of each cube. Maybe we should have this for fixing them for long
time experiments. Please have a look here:
https://github.com/bionanoimaging/UC2-GIT/tree/master/CAD/EXT_Alignment_Tool

Best

astraw, 2020-01-17:
We have a Prusa i3 mk3. The template is a good idea for holding things more steady. I think I would just physically join the two pieces to be one piece at the 3D model level to hold it really steady.

Thanks again.

beniroquai, 2020-01-22:
So all good for now? :)

astraw, 2020-01-22:
Yes, thanks.

closed 2020-01-22 by astraw