openUC2/UC2-GIT #26: Improvement of the TUTORIALS

b286, 2020-05-04:
We want to be user-friendly but to improve on that, we need users ;) 

Start with the UC2 project by following our [Tutorials of the TUTORIALS section](https://github.com/bionanoimaging/UC2-GIT/tree/master/TUTORIALS#-tutorials). Suggest improvements to the Tutorials if the instructions are not clear enough, more images are needed, something is not as obvious as we thought...

Mzyetik, 2022-05-27:
Tutorial-3 Lens Holder stl Files
STL file download options are not available on the links from the page https://github.com/openUC2/UC2-GIT/tree/master/CAD/ASSEMBLY_CUBE_Lens. For example , on the following link https://github.com/openUC2/UC2-GIT/blob/master/CAD/RAW/STL/20_Insert_Lens_holder_v3_-50mm.stl there is no download option.

beniroquai, 2022-05-30:
Hey @Mzyetik, thanks for reporting the error. We moved all STLs to this folder: https://github.com/openUC2/UC2-GIT/tree/master/CAD/RAW/STL

beniroquai, 2022-06-03:
Did you try hitting the `RAW` button? :-)

Alternative tools to access files easier through the browser are for example 
https://github1s.com/openUC2/UC2-Git

Mzyetik, 2022-06-03:
:D yes now I got there i got the directly the code to print as
[image] then but when I right click on RAW and save as, it let me to download, many thanks :)