openUC2/UC2-GIT #30: Clarification of Baseplate connectors

AlecVercruysse, 2020-05-28:
Hello,

What is the purpose of the [Assembly_Base_Connector_4](https://github.com/bionanoimaging/UC2-GIT/blob/master/CAD/ASSEMBLY_Baseplate_v2/STL/Assembly_Base_Connector_4.stl)? Is it just the 4u version of [00_Connector_Baseplate](https://github.com/bionanoimaging/UC2-GIT/blob/master/CAD/ASSEMBLY_Baseplate_v2/STL/00_Connector_Baseplate.stl), meant for 3D designs? [The readme](https://github.com/bionanoimaging/UC2-GIT/tree/master/CAD/ASSEMBLY_Baseplate_v2) describes it as "length of 4 baseplate units" but nothing more.

 I am asking because, if I was constricted to a 120x120mm print bed, but wanted a 4x2 baseplate, I would need to connect two 2x2 prints. I know am able to construct the 2x2 base using [10_Base_v2](https://github.com/bionanoimaging/UC2-GIT/blob/master/CAD/ASSEMBLY_Baseplate_v2/STL/10_Base_v2.stl). Is the purpose of the holes along the x and y axes meant to facilitate the connection of these baseplates, with m3 screws and nuts?

Additionally, what is the naming scheme behind your CAD files, particularly the two digit number in front? 

Thank you for the help!

beniroquai, 2020-05-29:
Hey @AlecVercruysse, 

thanks for getting back to us! The Connector "Assembly_Base_Connector_4" is indeed an adapter to tilt the baseplate by 90° in order to build in the 3rd dimension. For concatenating base-plates you can conveniently print 2x2 plates and screw them together with M3 screws and/or mount them on a massive baseplate (i.e. Thorlabs optical tabels/breadboards, wooden plates, etc. ). Does this makes sense?

The naming convention - perhaps a bit confusing - is summarized in our [MDK](https://github.com/bionanoimaging/UC2-GIT/tree/master/MDK) with a detailed figure [here](https://github.com/bionanoimaging/UC2-GIT/raw/master/MDK/IMAGES/UC2-Module-Developer-Kit0.png).
We tried to differentiate between external and internal components.

AlecVercruysse, 2020-05-30:
This makes sense, thank you for the clarification!

AlecVercruysse, 2020-05-30:
This makes sense, thank you!

I see that 10_ corresponds to the "cube frame" components, 20_ to the cube inserts, and 00_ to any external parts, but I do not see in the figure the meaning of items labeled with 40_.

Particularly, in your [flex z-stage](https://github.com/bionanoimaging/UC2-GIT/tree/master/CAD/ASSEMBLY_CUBE_Z-STAGE_v2/STL), how does the `40_XY_Stage_Clamp_Slide_9.stl`  fit into this scheme? It seems like the rest of the hardware that clamps the slide is labeled with `30`, which makes sense as it can be considered "external module" hardware.

beniroquai, 2020-05-31:
Well, thats a very good question 😊 
It might be possible, that this is due to a wrong naming convention. Sometimes things are lost in chaos. The best would be to have some kind of online-configuator which outputs all files necessary for one setup as done in the [openflexure](https://ofm-stl-selector.netlify.app/#/v6.1.2?enable_smart_brim=true&reflection_illumination=false&optics=rms_f50d13&camera=picamera_2&use_pilens_optics_module=false&pi_in_base=true&base=bucket&riser=sample&include_actuator_drilling_jig=false&motorised=true&use_motor_gears_for_hand_actuation=false)

AlecVercruysse, 2020-06-02:
Such a tool would indeed be helpful in the future. Thanks, I'll close this issue since I understand the baseplate clarification.

closed 2020-06-02 by AlecVercruysse