openUC2/UC2-GIT #31: Finishing MiniBOX

b286, 2020-06-03:
- [ ] eyepiece: yes or no? 
- [ ] sample holder - old version or new one with screws+magnets? 
- [ ] amount of baseplates: planned 8, do we want to sandwich it? -> then 16 
- [ ] sample mounting in smartphone microscope: is it clear and okay now? 
- [ ] lens holder: does the new version work well, stays closed? 
- [ ] which spring length/diameter for the z-stage? 
- [ ] z-stage working well?

RainerHeintzmann, 2020-06-03:
Is there an idea, where we finally get a low-priced set of lenses from?
Best wishes,
  Rainer

beniroquai, 2020-06-06:
Is there a way to add a simple spectrometer with the exsiting parts?

beniroquai, 2020-06-09:
Some more thoughts:
```
### Documentation

- Clear separation of the theoretical and experimental part 
--> Maybe: ray path, experiment ("template"), theory and resolution (with expected result)
--> "The phenomenon is in the foreground"
- Amount of text is ok; do not overburden pupils, keep up the "play instinct". 
- Text: In ray optics, it is better to speak of a "model"; e.g. "Rye-optics uses a model in which light rays are represented as lines..." 
- initial foil showing and naming all parts in the box (miniBox) (á la packing slip)
- Lens specification: "converging lens", diverging lens, focal length
- introduce the variables (e.g. f'=focal length, F'=focal point, a'=screen width, etc... according to DIN, see Pedrotti)
- Illustrated instruction for assembling a cube is missing
- QR codes with reference to the website / YouTube link?
- "Template" and lens position do not match, therefore reality! =Documentation (critical?)

###  Parts

- Haptic feedback per click is very desirable (this is lost for the time being with injection moulding?); metal stop is missing
- How can we mark the parts and their optical characteristics (colored, impact stamp?, adhesive dots)
- The box needs appropriate tools (hexagon) 

### Experiments

- It would be desirable if we could also build a spectrometer with the box (grating + slit) - should be feasible, right?

beniroquai, 2020-06-09:
And some input from Christina:

In your instructions there are various didactic elements which - according to my suggestion - should be clearly distinguishable graphically to facilitate understanding.

 ## Theory section

- illustrations - partly with explanations - in the form of beam paths
the images should be labeled again individually (converging lens, diverging lens, Galilean telescope...), especially if there are several images on one sheet
- Definitions ("Focus...")
- Mnemonics ("A telescope is...")

## Experiments

- Test setups
- Task (instruction for action and observation) (sometimes)
- Questions such as "What does the image look like through the lens?"
- Solutions, e.g. "With the right lens at the right distance..."

 

The structure should be the same for all instructions. Whether first theory part or first experiment is a matter of opinion, we discussed this before. Terms have to be defined beforehand, but we have also had good experience in explaining the observed phenomena AFTER the experiment.

 
The experimental part should also be uniformly structured, e.g. in the form question - structure - task - evaluation. The evaluation can be e.g. a small calculation, a gap text or similar... . Resolution can also be included, or you can make a sample solution - the teachers are always happy about that - but it should then be separate from the student worksheet.

One more tip about the tasks. They usually consist of an instruction for action ("Change the distance...") and an observation instruction ("Observe the object through the lens"). The latter can also be formulated as a question ("What does the image through the lens look like?"). The more concretely both are formulated, the better. "Test the lenses", for example, is very unspecific because the task of action and observation is not defined. A good test is always to have non-specialists read over it.

b286, 2020-06-12:
Include eyepiece: [10×/20](https://de.aliexpress.com/item/32897739739.html?spm=a2g0x.search0306.3.10.55b33e3cqAk1OO&ws_ab_test=searchweb0_0,searchweb201602_0,searchweb201603_0,ppcSwitch_0&algo_pvid=d07eadd7-1b43-443e-9a26-70fe7014a047&algo_expid=d07eadd7-1b43-443e-9a26-70fe7014a047-4) - 10,40€ per piece

b286, 2020-06-15:
The final designs of baseplates must be symmetrical for stacking and magnet-comptible. 
There are 8 baseplates in the miniBOX (final amount)

b286, 2020-08-05:
* making the IM stuff compatible
* baseplate puzzles

beniroquai, 2020-08-05:
Instead of the eyepiece we should go for 2*40mm lenses to build a Ramsden Eyepiece:
[image]

Since the cubes will be easy to open this is a nice alternative.

beniroquai, 2020-08-05:
We want a flyer which has a minimum dimension of 20x12mm (wide side is more important!) 
We want to have a Buchschraube which binds it. Like this:
[image]

b286, 2020-08-05:
Lenses: 
* 2× 40mm - [Artikel 2120](https://optikbaukasten.de/)
* 1× 100mm - [Artikel 2004](https://optikbaukasten.de/)
*1× -50mm - [LC1259](https://www.thorlabs.com/thorproduct.cfm?partnumber=LC1259)