openUC2/UC2-GIT #33: Inventor Files for Z-Stage? (and other modules)

AlecVercruysse, 2020-07-02:
Hello,

Will the inventor files be released for any of these modules? Converting an STL to a solid body makes for a generally unconstrained and tough file to edit.

I'm asking because I need to design an objective holder for the z-stage that fits the objective we use, and I would like to simulate the whole assembly to ensure alignment.

Thanks!

b286, 2020-07-02:
Hello Alec, 

The files shall be made available. It just takes time to export everything in a nice way, especially from Inventor. Also, many modules (and the Z-stage most of them all) are being actively improved and change quite often, so we only release the versions we consider reliable. 

The latest version of the Z-stage is one that we are happy with and I'm planning to have it completely documented in git by the end of July. 

Thank you for raising the issue

AlecVercruysse, 2020-07-02:
Thanks, I eagerly await its release!

closed 2020-07-02 by AlecVercruysse

beniroquai, 2020-07-15:
All files can be found [here](https://github.com/bionanoimaging/UC2-GIT/tree/master/CAD/RAW).