openUC2/UC2-GIT #35: XY stage

scientistnobee, 2020-08-04:
After reading your paper, I ordered the XY stage from Aliexpress. Along with the stepper motor connectors it has extra electronics parts, which I believe or end stops. Could you please confirm what they are for. The flex ribbon cable seems a better value, so can we find a connector for this?

beniroquai, 2020-08-04:
Hey @scientistnobee thanks for reaching out! I appreciate your interest in the UC2 project. The xy-stage from aliexpress indeed has flex cables, but buying the correct adaptor is quiet cumbersome. I didn't finde one which fits and designing a customized PCB makes reproducing by people outside the electronic community a little bit more complicated. 
Indeed there are two end stop sensors which can be used to detect the final position. If you're keen to use that for a calibration I would love to see that! Right now we do that with a raspberry pi camera and track the movement of the stage with sub-pixel frame registration. that works quiet well.

scientistnobee, 2020-08-04:
Thanks for your prompt reply and confirming that they are end stops. Do you have any link or extra info on how to use them. It has four solder joints. The sub pixel frame seems interesting. 

Some questions on the stage. Can a single XY-stage can bare the weight of a slide? Also, do we really need an extra IC to control the stepper motors? I think current from an Arduino board IO's is sufficient, so adding diodes might be enough to drive these tiny steppers.  

I can design the PCB as its my hobby. May be some people will be interested in it. 

Thank you.

beniroquai, 2020-08-04:
Hey, I found [this](https://www.hackster.io/news/these-two-small-steppers-are-one-giant-leap-for-maker-kind-7a6b3599cfae) a good read. I digged into the Aliexpress djungle, but unforutenyl I didn't find any additional information. 
Microstep scanning is more experimental. you need a second raspi lens and do a retro imaing configuration. From there you can do image processing. It's a 1:1 imaging in the range of pixelsize. Hope that helps :)

scientistnobee, 2020-08-04:
I think I come across the same project in a different version. https://youtu.be/s-a9CgvyQM8

I don't know much about the microstrip scanning. I will read about it soon.

beniroquai, 2020-08-05:
That's brillant! Thanks for sharing! :)