openUC2/UC2-GIT #40: UC2 needs a better way to hold objective lenses - with different threads

beniroquai, 2020-08-10:
Most cheap optics are with RMS threadings, but I guess we should make this somehow adaptable - at least the part which slides through the z-stage. @b286 do you have a good way of doing that? Maybe something in Thingiverse? 

Perhaps it should look more like this rather than a full radial hole? 
[image]

@b286 any idea?

RainerHeintzmann, 2020-08-10:
... by the way. We should somewhere have our own RMS Tap.

Ich habe gerade privat einige 6mm Stangen (Thorlabs compatible?) geordert.
Mal sehen ob die Magnetisch sind. Hab da so ne Idee ....

VG,
  Rainer

*******************************************************************
Rainer Heintzmann
Leibniz Institute of Photonic Technology, Albert-Einstein Str. 9, 07745
Jena, Germany
(preferred mail address)
Tel.: +49 (0) 3641 206 431 Fax.: +49 (0) 3641 206 139
Institute of Physical Chemistry, Friedrich-Schiller-Universität Jena,
Helmholtzweg 4, 07743 Jena, Germany
jenaphotonics®
Tel.: +49 (0) 3641 948 350

beniroquai, 2020-08-10:
What do you mean by tap? Could you provide any drawings?

Regarding the rails: we had something in our very first designs in 2017.

RainerHeintzmann, 2020-08-10:
Gewindeschneider

---------------------
Rainer Heintzmann
Institute of Physical Chemistry, Friedrich-Schiller-Universität Jena,
Helmholtzweg 4, 07743 Jena, Germany
Leibniz Institute of Photonic Technology, Albert-Einstein Str. 9, 07745 Jena,
Germany
Tel.: +49 (0) 3641 206 431
---------------------------------

b286, 2020-08-18:
So we should offer the options RMS / M25. Is there anything else commonly used?

beniroquai, 2020-08-18:
Yes. Mostly RMS and M25. Eventhough both have almost the same thread. Maybe
we could get inspired with the openflexure triangular mechanism for threads?