openUC2/UC2-GIT #41: Allow to mount 50mm diameter lenses

RainerHeintzmann, 2020-08-13:
Many of the easily available lenses (e.g. form Ajax Scientific have 50 mm diameter. It would be useful to also be able to mount them. 
They just about fit in the standard cube, but a different mount would be needed. It is not so important to have to full size as a transmitting surface.
Can one make a design for a variable holder for big lenses?

In addition it may also be interesting to make a design for a holder that can even deal with 10 cm lenses at a large distance from the front cube, for example to build a telescope with a 10cm entrance lense.

beniroquai, 2020-08-13:
Sounds interesting. We could definitely make a lens holder for the big lenses.

Something like this:
[image]

Can you please add more information about the lens diameter and thickness? 
A generic 50mm lens holder can be found here:
[20_Lens_holder_DIA_50mm.stl.zip](https://github.com/bionanoimaging/UC2-GIT/files/5070951/20_Lens_holder_DIA_50mm.stl.zip)

Would you mind to provide a drawing for the in-front-of-the-lens idea? That would help to clarify it a bit.

RainerHeintzmann, 2020-08-14:
That's pretty good. A retainment ring, which has the same size as your aperture and is flat and squishy would be good.

beniroquai, 2020-08-17:
Certainly. Would you mind to print this and see if this works? If so, I'll
provide the retain ring. We need to come up with an adjusted design though.
What are the lenses thicknesses?

RainerHeintzmann, 2020-08-17:
Here are the approximate measures:
(Double convex),  diameter= 50 mm, d_center=4,9mm, d_side=1.9mm
(Double concave), dameter=49,0mm, d_center=, d_side=4,8mm
(Convex-Concave, diameter=48,6mm, d_center=3.1m
( Concave -Convex), diameter=48,4mm, d_center=3,8mm
(Plano-concave), diameter=49,2mm, d_center=4,5mm
(Plano-convex), diameter=49,1mm, d_center=4,7, d_side=2,3mm

beniroquai, 2020-08-24:
Please find the new version of the lens adapter here:
[20_Lens_holder_DIA_50mm.zip](https://github.com/bionanoimaging/UC2-GIT/files/5120546/20_Lens_holder_DIA_50mm.zip)

If printed correctly it looks like this:
[image]

@b286 do you think one can add this design to the Thingiverser?

b286, 2020-08-25:
Making it adaptable to the lens thickness? That shouldn't be difficult, I'll make the openSCAD design for it.

closed 2021-02-03 by b286