openUC2/UC2-GIT #42: New puzzled Baseplate should ship with variable Holesize for different printers

beniroquai, 2020-08-13:
We should provide different holes for magnets to give people to adjust it according to their (non)-calibrated printers. Maybe a Thingiverse configurator could do the job - or just a list of different STLs. Since we reduce it to only one part rather than ```1x3```, ```2x5```, etc. we don't need to do that for all the baseplates.

b286, 2020-08-18:
I don't quite understand what you mean. 

We could have an openSCAD design of the baseplate and using the Thingiverse Customizer, one could change a single parameter (hole offset) to change the diameter of the holes for the magnets. 

However, this would require everyone who prints the baseplate to experimtn with this and find the optimal setting. If someone want just 8 baseplate units, they might find it easier to file the holes a bit. For a little number of part, I certainly would. 

But is this something you had in mind?

beniroquai, 2020-08-19:
That's exactly what I meant. The new baseplate has two parameters: whole
for the screw and the magnet. I guess this could be made adaptable. I had
to experiment quite a lot in order to make it fit perfectly.
Maybe worth a try? Don't know how much effort that means for translating it
to openSCAD.

Thanks! )

b286, 2020-08-19:
The effort to translate it to openSCAD is not such an issue. What I see as an issue is that every user who would print the baseplates would also have to experiment with the printing quite a lot. But if it simply doesn't work otherwise, then it's worth it.

beniroquai, 2020-08-24:
Hm. Maybe we should provide a print-template which has holes ranging from 4.8-5.2 in 0.1 mm steps. Users can print it and see how "precise" their printer do their job and then they can print the "real" one.  Something like[this](https://www.thingiverse.com/thing:242437)

This emphasizes the reproducibility aspect. 

For the base plate, the only parameters would be the hole-diameter for the screw (M5) and the magnets. Eventually also the dimension of one plate element which may differ and add up with combining multiple of them..

What do you think?

CC: @RainerHeintzmann @ranranking

RainerHeintzmann, 2020-08-25:
Sure. Sounds like a sensible approach. 
Best wishes,
Rainer

---------------------
Rainer Heintzmann
Institute of Physical Chemistry, Friedrich-Schiller-Universität Jena,
Helmholtzweg 4, 07743 Jena, Germany
Leibniz Institute of Photonic Technology, Albert-Einstein Str. 9, 07745 Jena,
Germany
Tel.: +49 (0) 3641 206 431
---------------------------------

b286, 2020-08-25:
UC2 printing template, sounds cool. 

Does it make sense then to consider more parts than just the baseplates?

ranranking, 2020-09-10:
About the template/resolution target stuff. I am thinking about, if it makes sense to use normal printer to calibrating their 3D printers. What i am saying, is try to create some of a template print with their own different 3D printers and kind of calibration target with thier normal paper printer. So can compare the difference of the printing quality. The thing i am not really clear is, if all the paper printer choose 300 dpi to print, how big tolerance will the printer bring onto the paper.

b286, 2020-09-10:
I've created a ciälibrating tool that works reliably, in my opinion :) It will go to github soon.

beniroquai, 2020-10-13:
Awesome! Thanks 😊

closed 2023-01-22 by beniroquai