openUC2/UC2-GIT #44: Add DOI badge to the github

beniroquai, 2020-09-07:
In case someone wants to cite this repo we need a DOI badge from our Zenodo archive or the biorxiv preprint. 

@b286 would you mind to take care of that?

b286, 2020-09-07:
Like this for the biorxiv preprint: 
[image] 

To get a DOI badge for the repository from Zenodo is a bit more difficult. We'd need to create a release and they kind of do not consider that someone wants to do this for anything else than software.

beniroquai, 2020-09-08:
<3!! 
Let's DOIt!

b286, 2020-09-09:
When you share the links to the badges that Zenodo created for our GITs, I'll add them all to the readme

beniroquai, 2020-09-21:
So the actual DOI has changed. It's this now:
```10.5281/zenodo.4018965``` 

Available here
[https://zenodo.org/badge/doi/10.5281/zenodo.4018965.svg](https://zenodo.org/badge/doi/10.5281/zenodo.4018965.svg)
[image]

b286, 2020-09-21:
Will you please also provide the DOI links? I've tried to search for it but it claims this doesn't exist.

beniroquai, 2020-09-21:
Would you also mind to add these two for the Hardware git:
[image]

and the Software git:
[image]

to places where you intended to have them? 

Thank you! :-) 
Now the code becomes citable

The DOI from zenodo becomes active once we publish it.

closed 2021-02-03 by b286

franciscozorrilla, 2021-04-21:
Hi I was wondering if you could share how you created the red bioarxiv DOI badge? Would love to make one for my preprint :)

b286, 2021-04-22:
Hi, 

I used the generic badge for GitHub as described [here](https://academia.stackexchange.com/questions/27341/flair-badge-for-arxiv-paper), in particular:

`[[image]](<doi link>)`

Using a colour that is somewhat specific for the journal or preprint service. B31B1B for bioarxiv. 

So for our preprint in bioarxiv `[[image]](https://doi.org/10.1101/2021.01.08.425840)` yields [image]