openUC2/UC2-GIT #45: Ordering Wastebox for recycling 3D printing material

beniroquai, 2020-09-25:
Long-time ago we were thinking about recycling 3D printing material. One service is this [one](https://zerowasteboxes.terracycle.com/products/3d-printing-materials-zero-waste-box?variant=1264347272&utm_source=www.terracycle.com&utm_medium=main_button&utm_campaign=ZeroWasteBox). 
I think that's fairly expensive. Maybe we can find some local alternative? 

Any ideas @RainerHeintzmann @b286?

b286, 2020-09-25:
I'm totally up for recycling the materials, the waste makes me sad. If we just use Prusament, maybe we could contact Prusa and ask if they would accept the material back? They produce everything themselves, they could have a way of dealing with this. I don't know :D

beniroquai, 2020-09-25:
Are you keen on doing that? :)

b286, 2020-09-25:
I can certainly try :D

beniroquai, 2023-01-22:
We have a plastic grinder. We will recycle stuff now with the new IM machine

closed 2023-01-22 by beniroquai