openUC2/UC2-GIT #49: Print Mechanical Z-stage without support or add supports?

dakoner, 2020-12-06:
This page:
https://github.com/bionanoimaging/UC2-GIT/tree/master/CAD/ASSEMBLY_CUBE_Z-STAGE_mechanical_v2
says the parts can all be printed without support.
I printed the focus linear bearing flexure:
https://github.com/bionanoimaging/UC2-GIT/blob/master/CAD/ASSEMBLY_CUBE_Z-STAGE_mechanical_v2/STL/Assembly_Cube_Z-Focus_Linearbearing_1x1_v4_20_focus_inlet_linearflexure_v4_7.stl

and while the piece prints OK without support in the default orientation, the print quality is a bit low (the long bridges have lots of extra squigglies).  I don't want to add support on the bridging flexures, since it would be tough to remove.  Is there a way to put some internal supports in that can be easily removed to improve those? 

BTW I'm using a Prusa MK3S with PETG, but I can also do ABS and PLA.

b286, 2020-12-07:
Hi, 

Thank you for your comment. We print this part with a 0,15 mm layer height with the Prusa MK3S, using various materials, and while there are some extra filament pieces where the printer prints 'in the air', this shouldn't have any effect on the performance of the part. 

If you feel like it'S needed, you can od support but removing it is not so easy and in my experience you risk breaking something that should stay in one piece.

Kirk3gaard, 2021-06-02:
Hi 

I am surprised how well this printed even though there are quite a few horizontal pieces that have to be printed in the air. 

Best regards 
Rasmus 

[image]

closed 2023-01-22 by beniroquai