openUC2/UC2-GIT #50: Lens holder for optical objective lens?

jediknightnapoleon, 2020-12-14:
Hi,

Just discovered this project. It is really amazing! Could you please let me know if it's possible to use the spiral or fixed lens holder for tube objective lens?

Many thanks!

Jerry

b286, 2020-12-14:
Hi Jerry, 

Thank you for your support :) The spiral holder can be generally used for any round object, the important parameter is the diameter of your tube lens. The holder found [here](https://github.com/bionanoimaging/UC2-GIT/blob/master/CAD/ASSEMBLY_CUBE_Lens_v2/STL/20_Cube_Insert_Objective_Holder.stl) can hold it if the diameter is up to 38 mm or so. It is useful to have the lens mounted in a tube or a ring. 

The fixed holders (for [small](https://www.thingiverse.com/thing:4377691) and [big](https://www.thingiverse.com/thing:4580156) lenses) can be adjusted for any diameter up to 50 mm but you need to have a bare lens in them.

jediknightnapoleon, 2020-12-14:
> Hi Jerry,
>
> Thank you for your support :) The spiral holder can be generally used for any round object, the important parameter is the diameter of your tube lens. The holder found [...]

Thanks a lot! I see, so I guess it would be better if I design a small ring/tube separately to mount the tube lens and then put the assembly on the spiral holder? Does the custom tube have to have any special structure to fit the holder? Also how does the spiral holder hold the lens securely? Does it only hold the lens horizontally or can it hold the lens vertically too?

b286, 2020-12-14:
If you don't have the lens in any mount, I'd rather suggest to use the fixed holder. the spiral holder is good for example for [mounted Thorlabs lenses](https://www.thorlabs.com/newgrouppage9.cfm?objectgroup_id=6277&pn=LA1131-A-ML) but as you can see in [this picture](https://github.com/bionanoimaging/UC2-GIT/blob/master/CAD/ASSEMBLY_CUBE_Lens_v2/IMAGES/CUBE_LENSHOLDER_1.jpg), putting in a bare lens means neccesarily touching it, which brings the risk of fingerprints and grease on the surfaces. 

The spiral is strong enough to just hold the lens in position, no matter whether vertically or horizontally placed in the cube. However, it might break after some time.

jediknightnapoleon, 2020-12-14:
Thanks, sory I didn't make this clear. I meant the microscopy objective lens like in [here](https://www.thorlabs.com/newgrouppage9.cfm?objectgroup_id=1044)
Is it ok if I use lenses like those?

b286, 2020-12-14:
I see, I understood tube lens. A microscope objective can be just placed into the spiral. However, in this case, it wouldn't be that stable when the objective is held vertically. For such application you can try the [new insert](https://github.com/bionanoimaging/UC2-GIT/blob/master/CAD/ASSEMBLY_CUBE_Z-STAGE_v3/STL/20_IM_Cube_Insert_Lens_holder_RMS_63x.stl) that is so far tested for the [new Z-stage](https://github.com/bionanoimaging/UC2-GIT/tree/master/CAD/ASSEMBLY_CUBE_Z-STAGE_v3). Would that solve it?

jediknightnapoleon, 2020-12-14:
I see, thank you. This seems promising. I'm just wondering how does this new insert hold the lens? Is it press fit or does it just have a smaller diameter so that the lens can rest on the hole?

b286, 2020-12-14:
It is pres fit - a RMS thread fits in the triangular hole in the insert.

jediknightnapoleon, 2020-12-15:
Do you have a picture of it in real life? It's a bit hard for me to imagine how this works. Don't worry if you don't have one. I guess I can just print one and try to sit a lens on :)

beniroquai, 2020-12-15:
Hey, certainly. I have just updated the documentation [here](https://github.com/bionanoimaging/UC2-GIT/tree/master/CAD/ASSEMBLY_CUBE_Z-STAGE_v3).
Please reach out if this is confusing. 
As @b286 already mentioned, we will switch to a new v3 design very soon. But don'T worry, everything stays compatible and great :-P

closed 2023-01-22 by beniroquai