openUC2/UC2-GIT #51: "Inner Cube" For parts which lack enough supporting material

beniroquai, 2021-01-04:
For parts like the AlliedVision camera, we should perhaps design inserts that are an inner cube rather than just a plate. This could help the part to not move around. 
What do you think @b286 ?

b286, 2021-01-04:
This would definitely improve the stability but also limit the alignment range. Personally I didn't have any trouble with the camera moving around but I did have to align it to a certain position. 

It would also increase the printing time of those inserts. 

We ca design something like this and try it out :)

beniroquai, 2021-01-07:
Hmh. The camera is not really stably mounted inside the cube. Perhaps this is better:
[image]

What do you think?
@ranranking

ranranking, 2021-01-07:
Can we maybe not use two walls here? Is it not good for cooling? I dont know if it will be better, have kind of inner post from the bottom of the camera plate to the 'roof', which direct contact with the four vertical pillar of the cube, and fix the plate.
Or just make the wall on top side as well and have a opening window, which can fix a 12v fan inside ^^

beniroquai, 2021-01-08:
Would you mind making a sketch or even a CAD drawing? ;)

beniroquai, 2021-01-09:
@ranranking made this:
[image]

It works very well. We should upgrade at least the camera insert with this. 
Can you please do that @ranranking ? replace the original version with yours and close this issue? Thanks!

ranranking, 2021-01-11:
I have saved the new version with camera in V3 folder with the name of Assembly_Cube_AlliedVision_Alvium_v3

closed 2021-01-11 by ranranking

beniroquai, 2021-01-11:
Cool! Thanks a lot :)