openUC2/UC2-GIT #54: UC2 configurator issues

b286, 2021-01-11:
### General issues  
- [ ] Lenses: how shall we take into account that there are different holders for different lenses and that the user might need to create his own? 

### GitHub 
- [ ] for v3 I omitted the version from the name of the module, since renaming everything everywhere is a hassle and this will save time in the future. However, all the modules that existed in git with ‘_v2’ in their name now appeared extra in the configurator and their folders also appeared again in the git CAD folder, with the only file in them being the ‘config.json’. The ‘_v2’ options in the configurator contain the v3 inserts that correspond to the module but not the cube and lid
#### STLs 
- [ ] is it necessary for each part to end with _v3? The .ipts and hence the generated .stls for which it matters do have it in their name already
- [ ] The renaming is a very good solution, although except for changing the prefix from UC2_v3 to UC2 only and cutting off the suffix number that Inventor creates when generating the file, I don’t see much difference. The “main” file names (without the prefix and suffix number) were matched between the excel and RAW/STL folder. Wouldn’t it be then enough to just cut the suffix numbers?

### Excel sheet 
- [ ] there must be some confusion between the v2 and v3 version of the excel sheet. or did you use a different tab than v3?
- [ ] the configurator doesn’t take into account the amount of parts stated in the excel sheet - shall the multiple parts be added as extra lines? 
- [ ] by making all the OPT parts default, the user might get more STL files than he needs - the question is whether printing some redundant parts makes him unhappy or whether it makes him unhappy not to find the parts in the zip folder when he needs them. I can’t decide what’s better 

### Configurator (general) 
- [ ] it would be nice to have a ‘clear all’ option for the case when one added a lot off different modules and got lost in it

b286, 2021-01-11:
### APPLICATIONS:
- [ ] all the downloaded zip files of the applications are empty, the list of files in zip is also always empty
- [ ] when adding a module of choice to an application, list and folder are again empty
- [ ] The applications sometimes give a different list of modules - the first time I selected an application there was a certain list. In some cases, from the second time on, there was a difference in either the first or the last module in the list. 
- [ ] The short descriptions of the setups that show under the selected app are the ones for a ‘complete overview’ tab, not from the latest ‘v3’ tab of the excel sheet

- [ ] Abbe experiment: 
- missing: 14x baseplate puzzle, 5x lens
- extra: 1x Sample holder

- [ ] Double-slit: 
- missing: 4x baseplate puzzle
- extra: 1x Sample holder

- [ ] In-line hologram: 
- missing: 4x baseplate puzzle
- extra: 1x RasPi Camera

- [ ] Incubator Microscope: 
- missing: 5x baseplate puzzle, 1x Z-stage
- extra: 1x Mirror 45

- [ ] Incubator Microscope fluorescence: 
- strange list, the APP should be anyways either updated or fused into the other one. Plus there’s another fluorescent microscope now and it became confusing 

- [ ] Light sheet: 
- missing: 15x baseplate puzzle, 1x Cylindrical lens, 2x Objective lens
- extra: 1x S-stage

- [ ] Mach-Zehnder: 
- missing: 11x baseplate puzzle
- extra: 1x Beam splitter 

- [ ] Michelson: 
- missing: 7x baseplate puzzle
- extra: 1x kinematic mirror

- [ ] Simple projector: 
- missing: 3x baseplate puzzle, 1x lens, (1xflashlight)
- extra:  -

- [ ] Simple smartphone microscope: 
- I should split it and make two list - for the mechanical-only one and the one with electronics
- missing: 8x baseplate puzzle, 1x Z-stage
- extra: 1x Sample holder 

- [ ] Simple telescope: 
- missing: 8x baseplate puzzle, 4x Lens (it’s empty)
- extra:  -

- [ ] Smartphone microscope:
- missing: 8x baseplate puzzle, 1x Sample holder, 1x Z-stage, 2x empty cube
- extra: 1x mirror 45

- [ ] Spectrometer: 

- missing: 6x baseplate puzzle, 1x Objective lens
- extra: 1x Sample holder

b286, 2021-01-11:
### MODULES:

- [ ] Baseplate 
- Baseplate 4x4 appers in the ZIP list, after clicking ‘download’ a bar saying ‘fetching data from GitHub’ appears but then nothing happens, nothing downloads 

- Alvium Cam ✓

- [ ] Aperture circular 
- everything works according to the list, but it’s supposed to have 7 leaves and not just one - shall Il add them as extra lines to the excel sheet or can this be fixed by selecting the amount from the right column?

- [ ] Aperture rectangular
- same trouble with the amount of parts that are supposed to be there more than once 

- Cube base ✓
- Basler cam ✓
- Beam expander ✓
- Beam splitter ✓

- [ ] Dichroic beam splitter 
- same trouble with the amount of parts that are supposed to be there more than once

- Eyepiece ✓
- LED ✓
- LED matrix ✓

- [ ] Laser 
- same trouble with the amount of parts that are supposed to be there more than once

- [ ] Lens 
- same as the baseplate: appears in the ZIP list, after clicking ‘download’ a bar saying ‘fetching data from GitHub’ appears but then nothing happens, nothing downloads

- [ ] Lens cylindrical 
- only insert, no cube 

- [ ] Mirror 45 
- nothing appears in the list, empty folder downloads 

- Mirror kinematic ✓
- Mirror kinematic 45 ✓
- Raspi cam ✓
- S-stage ✓

- [ ] Sample holder 
- it’s actually the sample comb 

- Z-stage mechanical ✓

- [ ] Z-stage objective 
- same as the baseplate: appers in the ZIP list, after clicking ‘download’ a bar saying ‘fetching data from GitHub’ appears but then nothing happens, nothing downloads 

- [ ] Z-stage sample 
- same as the baseplate: appers in the ZIP list, after clicking ‘download’ a bar saying ‘fetching data from GitHub’ appears but then nothing happens, nothing downloads

- [ ] Z-stage v3 
- works but the organisation of the Z-stage names is different in v3 now 

- two or more modules that work properly on their own can be added to the same selection and downloaded together, there are the correct parts in the zip folder

beniroquai, 2021-01-12:
Closing it for now, since all issues have been resolved here [https://github.com/bionanoimaging/uc2-configurator/issues](https://github.com/bionanoimaging/uc2-configurator/issues)

closed 2021-01-12 by beniroquai