openUC2/UC2-GIT #55: Adding Folder for Sample Preparation?

beniroquai, 2021-01-21:
Perhaps we could add a short list of recipes for sample preparation with images for those who have never dealt with this. 
@RainerHeintzmann you could give an example of preparing the flowers for example
@ranranking  @b286 - perhaps some cools stuff from your work? Cheek cells and fibers are nice samples too. 

What do you think?

darwinbandoy, 2021-01-29:
would be interested to contribute infectious disease (DxBox) module focusing on common diagnostics like fecalysis and maybe generate identification keys for common parasites in domestic animals ( my DVM background).

beniroquai, 2021-01-31:
Sounds great! I could simply "copy+paste" the immuno fluorescent stuff I did so far. But I think it would be great to have something for the "ordinary" hacker who tries a light sheet microscope for the first time. In best case we could have it as a recipe based on off-the-shelf components.

closed 2023-01-22 by beniroquai