openUC2/UC2-GIT #56: Magnetic screw issue (needs further documentation and alternative sources)

darwinbandoy, 2021-01-23:
First I would like to express my gratitude for the team who made this modular microscopy possible.

I was inspired to purchase a 3d printed to build my raman microscope. I have my spectrophotometer prototyped using lego blocks and on my way 3D printing the microscope components.

I purchased stainless steel (M3-0.50 x 12mm) on Amazon and the biggest surprise is it is not magnetic. This might be a potential issue for future makers, so can we add a list of tested and suitable materials specification indicating they can be magnetized? I see in the site a link to a European source but it would be useful to detailed out specific metal characteristics and warning that some stainless steel and zinc coated screws cannot be magnetized effectively.

b286, 2021-01-25:
Dear @darwinbandoy , 

Thank you for your nice words and your interest in the project. 

I believe we say that the screws should be made of galvanized steel in (hopefully) all the parts lists that appear through the repository. Stainless steel is in deed not magnetic. However, your point is very good - this might be causing confusion and leave the makers thinking that "steel is a steel, how different could it be". I will add an explanation to the Resources and the relevant pages.

dakoner, 2021-01-29:
'galvanized' is the not the best term to use if  you  search for  magnetic screws on amazon.  it brings  up a  lot of  construction screws  as well as machine screws.  galvanization is a surface coating, instead I think you want to select the steel's material properties 

Amazon actually has a "Screw filter" https://www.amazon.com/s?k=steel+screw&ref=nb_sb_noss_1 where  you  can  select  "carbon steel" a nd  deselect "stainless  steel".
I ended up getting https://www.amazon.com/gp/product/B015A34BUQ