openUC2/UC2-GIT #57: UC2_v3 release

b286, 2021-01-25:
When do we merge the v3 branch into master? What is left to be done before that? 

- [ ] We should keep a link to the pre-merge point of the repository history somewhere on sight, at least for some time, for people who might still be working with v2. 

- [ ] The video updates by @edabngl can be done after merging the branch 
- [ ]  The SIM+ISM documentation @ranranking What's the state?

Anything else?

beniroquai, 2021-01-25:
We can create a Tag or release to the current version as v2.1?

beniroquai, 2021-01-25:
@ranranking do you need help to move openSIM/ISM to the configurator? I think it needs some polisihing ;)

ranranking, 2021-01-25:
I will try to update the ISM and get a nice image for it as well, we can see if it need some polishing after my push

获取 Outlook for iOS<https://aka.ms/o0ukef>

b286, 2021-01-25:
- [ ] @b286  check and update the google spread sheet, hide confusing old information 

I think such a release wouldn't be that different from the last one. I'd suggest to put somewhere visibly in the main readme a link to the fixed point in the history of commits - the way we did it with the link in the Nature paper. That way it can be easily browed.

beniroquai, 2021-01-25:
You should see it here: 
[image]

Regarding ToDo's .. I'm waiting for the go from @ranranking then I'll review the UC2-configurator once again :)

ranranking, 2021-01-25:
I have pushed the new version of the  configurator excel, maybe you can have a look when you have time. I have upgraded the ISM module with the Z-Stage, STL and cover images are also in the folder they should be.

beniroquai, 2021-01-25:
Thanks @ranranking I checked the list. It compiles now, but there are issues with the parts it seems..

Also @b286 and @ranranking the following parts are still missing in the STL RAW folder: 
```
20_Insert_Large_Lens_holder_v3 is missing
30_LS_Sample_stage_v3 is missing
20_focus_inlet_linearflexure_motorized_small_v3 is missing
30_CUBE_openSIM_Lid_v3 is missing
```

Cover image for SIM is missing

beniroquai, 2021-01-25:
Looks good so far, except the ISM:

[image]

Intentially? 50 puzzle pieces? ;)

ranranking, 2021-01-25:
I added 50 pieces because i am not sure about how to configure it. I set 50 is for both openSIM and ISM togther, because in the tutorial i wrote all based on the setup combined togther but on the figure is separate. Should just write as on the example figure?

beniroquai, 2021-01-26:
That's fine; We need to decide if we really want to include all modules in the setup, but since the `APP_openSIM/ISM` is a fully assembled setup, it should have all the modules. Would you mind adding the other components too? I think you can copy/paste the entries from openSIM. 

Also once again, the spelling must (!) be the very same as the actual files ;-)

b286, 2021-01-26:
I know what the release looks like, but then you can just download it as a zip file. Considering the comfort of, for example, my currect high school pupils I think the option to browse through the page normally is more convenient. 

Missing files added. All files are (fingers crossed) the latest version.

beniroquai, 2021-01-27:
So then let'S create a new v2 branch derived from the last master before merging to v3?

beniroquai, 2021-01-27:
The latest commit looks good, no missing `STLS` any more thank you! 👍

beniroquai, 2021-01-27:
Also; The openSIM and openISM Application was still wrong, something was bad with the numbers..all shifted. Corrected it.

b286, 2021-01-28:
v2 branch, that's a neat solution! Thanks for the idea, I'll do that.

b286, 2021-02-03:
Anything missing or are we ready? 

Videos by @edabngl are almost all published now. 

If the SIM and ISM parts are done, I'd pull it into master this week.

edabngl, 2021-02-03:
Yes. I uploaded 6 new YB videos for v3 cubes and baseplates.

beniroquai, 2021-02-03:
I like what I see! Very much indeed! :) 
Well done @b286 and @edabngl!! 
I checked the configurator once more, everything looks good to me. Certainly there are some issues in the github, I'll try to sort these out soon! :)

You have my blessing :p

beniroquai, 2021-02-03:
@b286 perhaps we should think about a "launch campaign" like [this one](https://twitter.com/OpenFlexure/status/1356632596172668934) ;-)

b286, 2021-02-04:
Totally. You mean something big scale?

beniroquai, 2021-02-04:
Well, you had this funny idea of filming someone assembling an ordinary cube and our injection moulded stuff with a stop-watch in the background. Something like that. Could be "BIG" wrt. new version 3.

b286, 2021-02-05:
Yes, we'll film this with Eda once I can come to work. Shall we then wait with the release a use this video to advertise it?

beniroquai, 2021-02-06:
Yes! Let's summarize the advantages of v3 in a twitter thread like the one from [openflexure](https://twitter.com/OpenFlexure/status/1356632596172668934)

closed 2021-03-19 by b286