openUC2/UC2-GIT #58: Spectrophotometer alignment

darwinbandoy, 2021-01-26:
I am working on a raman microscope, one of the components is a spectrophotometer.

I decided to follow the current working guide but doing so will not generate any spectra.

The camera or imaging module must be perpendicular to the diffraction grating. 

In my case I placed a perpendicular 45 degree mirror so spectral image is reflected on top where a phone or webcam imager can be conveniently placed.

beniroquai, 2021-01-26:
Hey @darwinbandoy, thanks for filing this issue. Do you want to directly use the [UC2 Spectrometer](https://github.com/bionanoimaging/UC2-GIT/tree/master/APPLICATIONS/APP_Spectrometer) for Raman imaging? This sounds tough, but I'm sure you have adapted your design already or? 
@edabngl has made some experiments with the spectrometer, perhaps she can help you?

darwinbandoy, 2021-01-26:
I made multiple modifications: 

1) I created covers but PLA is too shiny so I added black paper to reduce reflection
2) I controlled light entry by coupling fiber optic as entry point
3) I designed a diffraction grating mount 
4) I modified the alignment of the diffraction grating with mirror and imaging system 

I am still doing some modifications but I can share my modified version if this valuable to the community. 

Would love to reach out to other makers out there.

beniroquai, 2021-01-27:
It would be brilliant to see that! I'm very curious. Would you mind sharing it here? 
Also if you want you can contribute with a pull request. 
We will ship the version "v3" in the next days. Everything will be compatible though, don't wonder ;-)

darwinbandoy, 2021-01-29:
So far the fiber optic spectrophotometer prototype is working (generates spectra). 

I will do some refinements and would share the results here.