openUC2/UC2-GIT #59: Having a low-cost version of a cylindrical lens

beniroquai, 2021-02-04:
Instead of having an expensive hard to get cylindrical lens, I tested a plane-convex plastic slab which is made out of Plexiglas (R) with an refractive index of 1.48 I suppose. You can get it very cheap at [ebay](https://www.ebay.de/itm/10-99-m-Acrylglas-Halb-Rundstab-15-mm-Hohe-7-5-mm-Lange-wahlbar/362697878979?hash=item54727a6dc3:g:WmAAAOSwUKxYm1H3)

This is the slab with a `radius=15mm`, so should give a focus of roughly `f'=15m`m`

[image]

Here it is hold in the holder:
[image]

And the `STL`-file
[20_Cube_Insert_Lens_Cylindrical_Plexiglas_glassslab_v3.zip](https://github.com/bionanoimaging/UC2-GIT/files/5924600/20_Cube_Insert_Lens_Cylindrical_Plexiglas_glassslab_v3.zip)

@b286 what do you think, should we have this as an additional module?

b286, 2021-02-04:
Definitely, "Cylindrical lens cheap". I'll create a page and excel input for it, so it will appear everywhere?

beniroquai, 2021-02-04:
I've added something to the excel. Will push it now

beniroquai, 2021-02-04:
You can have a look @b286

b286, 2021-02-05:
Do you want to replace the old cylindrical lens completely? 
Otherwise I'd just insert it as a separate module in the excel, with a place holder folder, explanations can be in the existing one.

beniroquai, 2021-02-05:
Sure go for it! )

closed 2021-03-19 by b286