openUC2/UC2-GIT #62: Magnetic balls pull force

jediknightnapoleon, 2021-02-17:
Hi,

Could you please let me know what's the minimum pull force needed for the magnetic balls? I saw that you used a magnetic ball that can hold 400g weights but that was produced in Germany and complicated to purchase from the UK after Brexit... So I was wondering if I have to buy my own magnets what pull force would be required?

Many thanks,

Jerry

b286, 2021-02-18:
Hi Jerry, 

I'm not sure what would be the required pull force but aside from that German vendor we also have good experience with ball magnets from eBay or Amazon. They don't provide such detailed specifications but in general the magnets always worked for us. So as long as you go for "strong neodymium ball magnets" you should be safe.

jediknightnapoleon, 2021-02-18:
I see, thanks!

closed 2021-03-19 by b286