openUC2/UC2-GIT #68: z stage for samples

jediknightnapoleon, 2021-04-08:
Hi,

I'm just wondering how does the z stage for samples hold the sample (for example a slide)? Do we just loosely put the slide on top of the translating stage? Also is it possible to just use a screw instead of a motor to adjust the height of sample manually like in the z stage for objective?

Many thanks,

Jerry

ranranking, 2021-04-08:
Hallo Jerry,

thanks for using the UC2 modules.

We have a new version of z stage, which moves the height of the sample instead of move the objective.
https://github.com/bionanoimaging/UC2-GIT/blob/v3/CAD/ASSEMBLY_CUBE_Z-STAGE_sample/STL/20_Cube_Insert_Z-Focus_single_v3.stl
It is designed for using a micrometre to fine adjusting the z position of the sample. You can get more information here:
https://github.com/bionanoimaging/UC2-GIT/tree/v3/CAD/ASSEMBLY_CUBE_Z-STAGE_sample

I hope it solves your problem and feel free to come back to us again.

Cheers,
Haoran

beniroquai, 2021-04-08:
Indeed, the Z-stage is a critical topic ;-) 
Always open for suggestions! In fact, the stage can be used for moving the objective as well as the sample. How do you like to work with it? Perhaps we can adapt it to make it more useful?

jediknightnapoleon, 2021-04-09:
Hi both,

Thank you very much for your replies! I see, could you please advise on how to assemble the z stage with a micrometer as the z controller? On the z stage cube page it only shows how to assemble with a motor so I'm not fully sure how a micrometer can be inserted. 

Could you please also recommend any specific micrometers that fit this version of z stage? 

Many thanks,

Jerry

beniroquai, 2021-04-17:
Hey, thanks for the remark. Indeed the Z-stage using the micrometer doesn't seem to be very well documented. 
We use micrometers from [RS Electronics](https://uk.rs-online.com/web/p/micrometers/7857878/).. @ranranking would you mind to add some images from your z-stage with the micrometer and the link to the RS parts please? :)

ranranking, 2021-04-21:
Hallo,

Here is a picture of the z stage with micrometre, the micrometer is fixed with screws from 3 sides and the rubber band can provide the backlash. 
[image]
It's better to use the similar micrometre as on the picture with a flat front surface on the rod, otherwise it will have a step leap after every turninig cycle.
An example is showing here
https://github.com/bionanoimaging/UC2-GIT/blob/v3/CAD/ASSEMBLY_CUBE_Z-STAGE_sample/IMAGES/Zstage.jpeg

jediknightnapoleon, 2021-06-02:
Hi would it be possible to just use a screw to move the z stage? I seem to remember there was a previous version which just use a long M3 screw but I'm not sure how it works exactly.

b286, 2021-06-02:
Hey, 
There is still the [mechanical Z-stage](https://github.com/bionanoimaging/UC2-GIT/tree/master/CAD/ASSEMBLY_CUBE_Z-STAGE_mechanical) that just uses a wheel on a long M3 screw. Other than that, I don't know which version do you have in mind.

jediknightnapoleon, 2021-06-02:
Hi,

Thanks for your reply. I tried the mechanical Z-stage you mentioned but had to modify the objective holder to fit my Olympus RMS20X objective. Unfortunately as this is bigger than yours the weight of the objective seems to cause the objective-holder complex to rotate around the screw and point slightly downwards. Is there any way I can fix this misalignment?

Many thanks!

b286, 2021-06-09:
Hi, 

I'm afraid we don't have a solution for this at hand, as we never used the mechanical stage for a real microscopy application, but rather for education with the small objective. However, we are currently working on a new, improved, and "stronger" objective stage, so a solution is hopefully coming soon.

beniroquai, 2021-06-09:
May I ask you to take a picture from your z-stage? PErhaps we have alternatives ;-)

jediknightnapoleon, 2021-06-14:
Hi all,

Thank you for your replies and sorry for the delay. I have attached two photos of my current cube for the objective. It basically uses the mechanical z stage but I modified the objective holder. It now holds the lens but if I try to adjust the position (e.g. translate the lens for focusing) of the objective then the orientation of the objective changes too because there is large friction between the holder and the 'fence'.
[image]
[image]