openUC2/UC2-GIT #73: Questions about Lenses and the Beam Splitter

Zmrf, 2021-05-21:
Hello,
I've got 3 Questions regarding the Fluorescence Microscope (infinity):

1. In "Parts" it says "Tube lense according to objective lens", what kind of lens is needed for an 10x objective?
2. The Laser, the Beam Splitter and the Emission Filter need to be choosen specific for the used Fluorescence Protein, right? 
3.  If yes, what Kind of Beam Splitter ist needed for TurboGFP with Excitation max at 482 nm and Emission max at 502 nm?

Greetings
Darius

beniroquai, 2021-06-04:
Hey Darius, 

many apologies for the late repyl! 

In order to answer your questions: 

```1. In "Parts" it says "Tube lense according to objective lens", what kind of lens is needed for an 10x objective?```

The tube lens in combination with the objective lens determine the full magnification of your optical system. For Zeiss, the tubelens length is 165, for Nikon/Olympos 200mm for example (not always though). If you grab an infinity corrected lens form Aliexpress or alike I would calculate with either 180mm or 200mm if it's not specified. More information are to be found [here](https://www.microscopyu.com/tutorials/offaxisrays). 

```The Laser, the Beam Splitter and the Emission Filter need to be choosen specific for the used Fluorescence Protein, right?```
Yes, exactly. If you want to have a filter cube for GFP for example, you excite the sample with 450nm Laser (blue), the protein fluoresces in green (e.g. 500nm) and this signal you want to detect on a camera. So you would need a system that reflects the blue, but does transmit the green. You can find more information on our [github page](https://github.com/bionanoimaging/UC2-GIT/blob/88be02412cd414ffad5aac8310d01578f0929374/CAD/ASSEMBLY_CUBE_Dichroic_Beamsplitter/Readme.md) and also from the people from the openflexure community [here](https://gitlab.com/openflexure/openflexure-microscope/-/issues/131), [here](https://github.com/rwb27/openflexure_microscope/issues/43) and [here](https://www.osapublishing.org/boe/fulltext.cfm?uri=boe-11-5-2447&id=429869) (also for GFP if I remember correctly). 
Let us know if you have more specific demands. We usually work with synthetic dyes (e.g. Alexa Fluor).

```If yes, what Kind of Beam Splitter ist needed for TurboGFP with Excitation max at 482 nm and Emission max at 502 nm?```

I guess so. 

Greetings

Zmrf, 2021-06-16:
Thanks for ur Answer.

Greetings :)

closed 2021-06-16 by Zmrf