openUC2/UC2-GIT #74: baseplate STL files for all offset values

Kirk3gaard, 2021-05-27:
Hi

Thanks for making such a cool project with open hardware.

I have been trying to generate STL files for the baseplate using the thingiverse customiser as described in the base plate puzzle guide. However, it appears to take forever for thingiverse to generate the new file. Have you considered generating the STL files for all the different off set combinations once and allow direct download for this?

Best regards
Rasmus

[image]

b286, 2021-05-27:
Hi Rasmus, 

I'm sorry for this issue, the Thingiverse Customizer tends to be buggy from time to time. You can also use directly the OpenSCAD file [here](https://github.com/bionanoimaging/UC2-GIT/blob/master/CAD/ASSEMBLY_Baseplate/3DP_Baseplate/10_Base_puzzle_v3.scad) and generate the STL file there. 

The STL file take a lot of space so we're rather trying to minimise the amount of the that is stored in GitHub. 

Thank you for your interest in the project,
B

Kirk3gaard, 2021-05-27:
Hi B

Thanks  a lot. The OpenSCAD file that seems to do the trick.

Best Rasmus

[image]

beniroquai, 2021-05-29:
Nice print! :) 
Which printer is this?

Kirk3gaard, 2021-05-29:
Hi 

It is the prusa mk3s.

The prints got even better after I discovered that the prusa slicer got an easy solution to get rid of the edge ("elephant foot" https://help.prusa3d.com/en/article/elephant-foot-compensation_114487 ) that formed at the bottom of the first few test prints :D. Now the puzzle pieces fit perfectly without sanding.

I assume other slicers have similar options. So it may be worth adding to the printer recommendations.

Best Rasmus