openUC2/UC2-GIT #75: The Openflexure Adaptor misses images and needs an update

beniroquai, 2021-06-07:
I just wanted to add this so that we won't forget. The [page](https://github.com/bionanoimaging/UC2-GIT/tree/master/CAD/CUBE_INSERT_OpenFlexure) misses images. Perhaps we can also update the parts? Are they up-to-date?

b286, 2021-06-09:
Images restored with latest commit, they got lost on the way. I don't think it's up-to-date, but I've never worked with it myself. Maybe @ranranking uses the latest version?

ranranking, 2021-06-09:
You mean the single adapter for Openflexure? I will try to find it and post image here, you can check if is that one

beniroquai, 2021-06-09:
That would be great. Then we can close this thing again :)

>

Tomvhattem, 2023-10-24:
I see there is only an adapter for the bottom. We also made an adapter to fit on the top, is that something we can add to this repo?

beniroquai, 2023-10-25:
Certainly! You can fork it and then file a pull request! That would be wonderful :)