openUC2/UC2-GIT #77: Cad scales in ISM and SIM STL's incorrect

Sh4zKh4n, 2021-08-15:
The raw STL files are scaled incorrectly for ism and SIM parts.

Could you let me know what the scaling should be.

ranranking, 2021-08-16:
Hey, i am sorry to make the confusion. The STL files are exported with Inventor and due to the incorrect setting, all the files are smaller than the real scale and it should be 1000% scale when you print it. I will also update the correct version as well. Thank you

Sh4zKh4n, 2021-08-16:
No Thank you, I was going crazy because I kept thinking. Why is this so small... 

Ill leave this open till you get a chance to repair but at least I can scale the thing! Much appreciated.

ranranking, 2021-08-16:
I update the STL files and you can find them here https://github.com/openUC2/UC2_openSIM/tree/main/CAD, https://github.com/openUC2/UC2_openISM/tree/main/CAD. They should now in the correct scale. I will close the issue now and if you have further questions, you can open an new issue

closed 2021-08-16 by ranranking

Sh4zKh4n, 2021-08-16:
Great thanks!