openUC2/UC2-GIT #78: Example Filters for Dichroic Beam splitter Cube

Micha-Mo, 2021-12-14:
Please give some example based on your used laser, how to choose the 3 filters according to wavelenght to use in the fluorescence microscope.

beniroquai, 2021-12-14:
Hey @Micha-Mo thanks for reaching out. 
What kind of fluorophore do you have if I may ask you? I think a good introduction into the selection of filters and light-sources can always be found [here](https://www.microscopyu.com/techniques/fluorescence/introduction-to-fluorescence-microscopy). 

If you have already know what to image, we can definitely help you selecting the filters - although the selection of low-cost filter dramatically reduced over the last few moths :-/
good sources: [comaroptics.com](https://www.comaroptics.com/components/filters/dichroic-filters/long-pass-dichroic-filters)

beniroquai, 2022-03-03:
Were you able to resolve the issue?
Let me know if you need further help.