openUC2/UC2-GIT #79: PSRAM ID read error - ESP32 while flashing LED matrix code

dhersm, 2022-01-05:
[image]

Can't connect LED matrix with the ESP32 for the Incubator microscope. While checking the serial monitor, it shows the error in the picture given.

errors:
PSRAM ID read error: 0xffffffff

abort() was called at PC 0x40089631 on core 1

beniroquai, 2022-01-06:
That is hard to debug. Did you enter the correct settings in the ESP32 Arduino IDE tool? Did you activate the PSRAM? 
Could you check just uploading the LED Matrix example from Adafruit?

dhersm, 2022-01-07:
Yes ,  the settings are correct. As you suggested, we enabled PSRAM and the PSRAM error is solved but "abort() was called at PC 0x40086ed5 on core 1"  is still there. The motor is connected successfully via MQTT but the LED array is not.

beniroquai, 2022-03-03:
I hope the error was resolved finally.

closed 2022-03-03 by beniroquai