openUC2/UC2-GIT #80: mechanical z-stage

Feanor007, 2022-01-23:
Hi all, it seems the inventor file for the mechanical z-stage is no longer available.
Does anyone know where to find it?

beniroquai, 2022-01-23:
Hey @Feanor007 good point. All a bit confusing a bit with the stages since they developed faster than expected. Would you mind which stage you're looking for? With motors? Which motor? (Nema/28BYI?) Flexure based or with linear bearing from CNC machines? 
I guess we should have a separate Github repo for a stable version of the stage!

Feanor007, 2022-02-01:
https://github.com/openUC2/UC2-GIT/tree/v3/CAD/ASSEMBLY_CUBE_Z-STAGE_mechanical
I was referring to this one, without motor

beniroquai, 2022-02-08:
Hey @Feanor007, sorry for late reply. 
I just updated the RAW IPT files. You can find them [here](https://github.com/openUC2/UC2-GIT/tree/master/CAD/RAW/IPT)

closed 2022-03-03 by beniroquai