openUC2/UC2-GIT #81: Motorised polarizer/filter rotating stage

GiuseppeCiccone96, 2022-02-11:
Hi,

I think it would be really nice to integrate a motorised polariser/filter mount that is compatible with the uc2 framework and easily adaptable to other open-source microscopy projects such as the OpenFlexure.
A starting point can be [This paper](https://opg.optica.org/ao/fulltext.cfm?uri=ao-60-13-3764&id=450413), where they present a step-by-step guide to 3d print and assemble motorised rotating mounts for both transmission and reflection configurations!

beniroquai, 2022-02-14:
Hey Giuseppe, 

thanks for pointing to this nice paper. I would really like to see this functionality in our system. Also it could help reducing speckles. 
I did a quick mockup using 3 ball bearings, a 3D printed timing belt and a 28byi stepper motor. What do you think?
[image]

GiuseppeCiccone96, 2022-02-14:
Hi Bene,

thanks so much for the quick try! 

I think it looks very promising - perhaps it could be integrated into the "APPLICATIONS" repository with instructions? 
Maybe one example could be an illuminator for polarization microscopy (I would like to try this out but we are currently moving labs and I don't have access to any equipment).

Thanks a lot for your support :) 

Giuseppe

beniroquai, 2022-02-16:
[Tadah](https://github.com/openUC2/UC2-Motorized-Rotator). 
[image]

Feel free to comment :-)

closed 2022-02-16 by beniroquai

GiuseppeCiccone96, 2022-02-16:
Hi Bene, looks fantastic! 

I think the link may be broken though?

Giuseppe

beniroquai, 2022-02-16:
Fixed. Sorry, for some reason it was set to private :)

GiuseppeCiccone96, 2022-02-16:
not sure why but it still gives me error 404... :(

beniroquai, 2022-02-16:
Hey, I'm very sorry. Now its really public!! ;-)