openUC2/UC2-GIT #82: UC2 injection moulded cube optimization thread

beniroquai, 2022-03-03:
I wanted to take the chance to collect some feedback in order to improve the injection moulded cubes. Please feel free to add whatever comment so that we will have a refined version that is less harmful, yet compatible and long lasting. 

I wanted to talk about this cube/baseplates
[image]

@edabngl, @b286, @renerichter, @ranranking

Some points I would like to revise:

- [ ] Sharp edges cause pain when pressing parts together or simply may cut the skin
- [ ] Where to place potential electronics?
- [ ] Sysytem is mechanically overconstrained 
- [ ] The inserts can never be "straight" since there is no kinematic mounting possible
- [ ] The current cube is not 3D printer friendly (e.g. the pins are not M5 or alike)

**Further important question:**
Do we want to keep the standard persistent or do we want to introduce a new formfactor?

I'm very open for any kind of discussion! Thank you!

beniroquai, 2022-03-06:
In order to get the cube "straight", perhaps we could think about a mix of the Thorlabs rails and the mysterious railoptics:
[image]

I could imagine having a modified version of an insert or even some extra hack at the cube's edges? 

Or even a modified thingy from the caliper. It's quite accurate to work with it

[image]

RainerHeintzmann, 2022-03-06:
That's not a bad idea. Let's talk about it.

beniroquai, 2022-03-06:
Sure!

One more thing to add: baseplates should be compatible with laser cutting
manufacturing.

ranranking, 2022-03-07:
I'd like to have some lock mechanism, it helps a lot to hold the optics position. Is that possible to have a small groove at cube parts connection that can open the cube easily without threaten the optics?

beniroquai, 2022-03-07:
One more "thing": make magnet-based and stackable option compatible by adapted baseplate, but same IM cube:

- have the holes for the pins in the baseplate bigger so that no force is required to put them together
- laser-cut parts could be used for magnet-based version? 3 layers, then magnets are "fixed" forever? (e.g. Sandwich)

beniroquai, 2022-03-07:
And: "expansion mechanism" for fixing the insert permanently.

beniroquai, 2022-03-08:
@ranranking do you mean a lock mechanism to not accidentally open the cube`? Definitely Important!

beniroquai, 2022-03-16:
One more thing I really don't like is that the baseplates are sometimes sticking together too stiff. Also the cubes are residing on the baseplate so that one cannot get them out easily. Sometimes the pins are too big for the holes which causes an additional distance between plate and cube. 

One more thing is the positive/negative wings of the puzzle piece. Sometimes they are "in the way". Solution would be a pure "female" and "male" version of the baseplate?

Kubiczek36, 2022-03-20:
Hey,

I like the simple way the cubes are. Maybe I would think about modifying the inserts. Here I got some suggestions.

[image]
[image]

And as i understood it right. You are aiming to produce the baseplates on laser cutter (for example from PMMA). And there would be software to generate the files for the cutter based on which layout you like to use. Then I think could be nice to make _two-story architecture_ with zero floor for electronics, stretch the cables through the holes in the baseplate to control the _optics floor_.
[image]

beniroquai, 2022-03-20:
That looks great! We should have a hackathon for brainstorming a bit! 
For locking I though a bit simpler. Similar to these "guys":
[image]

Having a flexure bearing that is squeezed and then locked in place of the cub - and potentially even straightens the insert. 

Electronics is a whole different thing. So far we collect some ideas [here](https://github.com/openUC2/UC2_Feather). This is cube-contained + a baseplate connector, but this is just a mock-up.  I did not fully get your mechanism yet. Night to play with the doodle.

RainerHeintzmann, 2022-03-21:
"Similar to" could possibly mean "use these guys"?

beniroquai, 2022-04-08:
We should also think about having an insert inside an insert, so that we can injection mould the global insert in order to have higher precision + better reproducibility. + the global insert should have a locking mechanism

beniroquai, 2022-04-08:
Having more holes for adding shielding plates is also mandatory

beniroquai, 2022-04-21:
The "Masterinsert" should have press-inserts for screws. Perhaps M3. PErhaps ferromagnetic?

beniroquai, 2023-09-29:
- Wechseleinsätze um Noppen rasuzunehmen
- Löcher an Seite genauso groß wie Puzzle (34mm)
-