openUC2/UC2-GIT #88: STEP missing

Kubiczek36, 2022-06-02:
Hey hey,

looking in the repository all the CAD files are there in Inventor file format, however not everybody uses Autodesk software. Please could you begin to add the files also in STEP as it is universal format, which can be opened in other CADs also?

My fellow student Veronika met this issue working on her thesi, whe she had to import the STL files into Solidworks. Trust me, you do not want to work with a mesh in CAD. However the biggest issue arrived when she wanted to make the drawings from the assemblies. There are too many lines creating the mesh which are not the edges normally figured into the drawing.

I know there is quite a lot of models however it would be good to begin to add the step at least for the most used models.

Thanks!

beniroquai, 2022-06-02:
Thanks @Kubiczek36 for reporting this issue. 

I found this:

[image]

Let'S see what will happen now :-)

Would you mind reminding me to upload the STPs soon?

b2jia, 2022-11-24:
I echo @Kubiczek36's suggestion! Having these files as STEP format would make modifications much easier. I believe it will also facilitate users to contribute to this repo - I am happy to commit my changed files!

beniroquai, 2022-11-27:
Hey! Thanks for putting this up again. I was looking through the
documentation of Autodesk Inventor and it seems that it's not possible to
have the STP files exported automatically in batch mode. Do you know any
way of doing that?

Message ID: ***@***.***>
>

beniroquai, 2022-11-27:
I managed to initiate a batch run - but where to store the files? The git
would be too large then.

Am So., 27. Nov. 2022 um 22:54 Uhr schrieb Benedict Diederich <
***@***.***>:

> Hey! Thanks for putting this up again. I was looking through the
> documentation of Autodesk Inventor and it seems that it's not possible to
> have the STP files exported automatically in batch mode. Do you know any
> way of doing that?
>
>
> Message ID: ***@***.***>
>>
>

b2jia, 2022-11-30:
Is it possible to host on Zenodo? 

Sorry I missed the last message, for some reason it did not appear on my notifications.

beniroquai, 2022-11-30:
@b2jia, please find the STEP files here: https://github.com/openUC2/UC2-STEP

closed 2022-11-30 by beniroquai