openUC2/UC2-GIT #91: About Fluorescence Microscope (infinity)

Mzyetik, 2022-07-17:
Dear All,

Some of the component links dont work (especially this one 1x Objective lens, No Name, 100x, NA1.25, 50 Euro, China ) in the Infinity Florescence Microscope page:

https://github.com/openUC2/UC2-GIT/tree/v3/APPLICATIONS/APP_Fluorescence_Microscope_infinity

We would be happy if it could be updated.

Besides, here we wonder about the objective lens. In this Florescence Microscope, do we need a special objective florescence lens? (which 10 times around more expensive). Or are the non-florescence objective lenses currently suggested proper also for florescence?
We highly appreciate your answer in case we may buy directly a florescence objective lens instead a normal one.

many thanks

beniroquai, 2022-07-19:
Hey, thanks for reaching out. 
In general the quality of the objective lens is determined by its name (e.g. apochromat). Fluo usually means that you have additional antireflection coating. Additional information is nicely summarized here https://www.microscopyu.com/microscopy-basics/introduction-to-microscope-objectives

Decent lenses for relatively low pice are available from Nikon and Bolioptics
e.g. https://www.bioform.de/shop.php?action=tree&wg=2&pid=5107&treeid=5308
and https://bolioptics.com/microscope-parts-accessories/objective/infinite/

Looking forward to your PR :)

Mzyetik, 2022-07-19:
Thank you also for your kind answer about Fluorescence 

How about achieving phase contrast in your setup? Do they need special objectives as we are used to?

many thanks

beniroquai, 2022-07-20:
Hey, you can use oblique illumination or darkfield. The alternative would
be purchasing a phase-contrast objective and use a ring illumination.

> Message ID: ***@***.***>
>

Mzyetik, 2022-07-26:
Many thanks for your kind answers, I hope we can mount it on UC2 microscope setups, by the way how to print focus_inlet_linearflexure https://github.com/openUC2/UC2-GIT/blob/master/CAD/RAW/STL/20_focus_inlet_linearflexure_v2-3_1x1_v3.stl 
[image]
we were not successful although did several trials , especially the support is not removable , we would be happy for your kind advice.

beniroquai, 2022-07-26:
Hey @Myzetik, 

thanks for your message. In fact, this stage is already fairly old and I cannot support this design anymore. We moved to NEMA stepper motors in our recent results. 
If you send a photo, perhaps I can tell what could be improved ;)

Mzyetik, 2022-07-27:
Thank you very much for your answer, and informing about the new version. I checked it as well https://github.com/openUC2/UC2-GIT/blob/v3/CAD/RAW/STL/UC2_20_Cube_Insert_Z-Focus_single_motorized_v3.stl regardless of new or old version, the 3D printer requires to use support to print these nicely designed structures but then it is not possible to remove the support as seen the one on the picture.
[image]
We are so curious how could they be printed (what is the trick to manage it with normal PLA printers, here we use Ultimaker 2+ . Many thanks for your support and patience :)

beniroquai, 2022-07-28:
Did you try printing without the support? Not sure if @b286 has any insights here ;)

wbrett87, 2022-08-15:
Hi, if you havn't resolved this issue yet:

1) Flip it over in your slicer software so the largest area touches the printing bed. 
2) Print WITHOUT support - this is what helped the most for me. Your printer should be able to "bridge" this distance.

beniroquai, 2022-08-15:
Thanks!:)
In general: if you use the full prusa ecosystem including their filament and slicer, chances of a successfully printed microscope are very high. :)

wbrett87, 2022-08-16:
I used a Creality 3 ender pro with good results

beniroquai, 2022-08-16:
Which settings and Filament if I may ask you?

wbrett87, 2022-08-16:
I used Inland blue PLA(from Microcenter here in US).  It needs hotter temps than your average PLA but once I figured that out it worked great. I can't remember the settings offhand but I will put them here when I get home from work. I used the Ultimaker Cura slicer.

Mzyetik, 2022-08-16:
> 

Thanks a lot, some of our friend used water soluble support material but it resulted with a warped holder, then the better solution is your suggestion we will try it, thank you again for the continuing discussion, it is quite informative