openUC2/UC2-GIT #96: FLU Modul for Smartphone Microscope

Mzyetik, 2022-09-05:
How to add Fluorescent Module  for Smartphone Microscope https://github.com/openUC2/UC2-GIT/tree/master/APPLICATIONS/APP_SMARTPHONE_MICROSCOPE  your advice and guidance are highly appreciated

beniroquai, 2022-09-05:
What kind of fluorescence would you like to see?

Mzyetik, 2022-09-07:
Initially, we just wanted to follow the steps and the flu module mentioned here in the link https://github.com/openUC2/UC2-GIT/tree/master/APPLICATIONS/APP_SMARTPHONE_MICROSCOPE , unfortunately we could not see the instructions neither the link of the flu led (faded away).

beniroquai, 2022-09-08:
Thanks again for reaching out. We are about to publish a new fluorescence
microscope based on this assembly: It's not up-to-date yet, but we will be
working on it soon!
The off-axis LED illumination gives a very poor SNR for the fluorescence
signal unfortunately. Depends a bit on what you want to do. :)