openUC2/UC2-GIT #100: Incomplete parts list for Polar BOX

b2jia, 2022-10-24:
Really interested in building up a PolarBOX for self-learning! I noticed that 

1. many of the links on [the parts list](https://github.com/openUC2/UC2-GIT/tree/master/TheBOX/PolarBOX) are either broken or missing (ie. which mirrors? which circular polarizers?)
2. the 3D parts are not on [UC2-configurator](https://uc2configurator.netlify.app/), is there a reason? 
3. assembly links are broken

Thanks in advance!

edabngl, 2022-10-24:
Hi. Actually, all STL files are included specifically in each application page, STL folders. But I recognised now that Build_Me page links need update, thank for feedback.

Just follow [this page ](https://github.com/openUC2/UC2-GIT/tree/master/TheBOX/PolarBOX)  and the application page links for STL fıles. Unfortunately, we don't have ready-to-download file for all necessary 3d printed parts but will add it in the following period. And the broken links will be fixed.

beniroquai, 2022-10-24:
@edabngl are you keen on fixing the links? ;-)

beniroquai, 2022-10-24:
I quickly implemented a URL-checker in github actions: https://github.com/openUC2/UC2-GIT/actions/runs/3312545528
I fear there are many dead links :-/

edabngl, 2022-10-24:
@beniroquai I can organize the folder, deletes unnecessary parts and add the links to already organized pages. But not this week, maybe next week :)

beniroquai, 2022-10-24:
Cool :)

Here is the summary:
https://github.com/openUC2/UC2-GIT/actions/runs/3312545528/jobs/5469341816

Look for "dead links" ;-)

b2jia, 2022-10-24:
Thanks both for your support. I want to clarify one thing - it looks like a few of the non 3-D printing parts are not listed (ie. mirrors, birefringent material, circular polarizers). Is there a parts list for these as well?

edabngl, 2022-11-17:
Hi @b2jia . I updated [BUILD_ME](https://github.com/openUC2/UC2-GIT/tree/master/TheBOX/PolarBOX/BUILD_ME) page of PolarBOX, broken links and missing images and links were added. If you can't find the details of the module in BUILD_ME, please go to directly the Application page for more details i. e. [Circular Polarization](https://github.com/openUC2/UC2-GIT/tree/master/APPLICATIONS/APP_POL_Circular_Polarizer) for the details of the circular polarizers and [Stress Birefringence Experiment](https://github.com/openUC2/UC2-GIT/tree/master/APPLICATIONS/APP_POL_Stress_Birefringence) for birefringent sample alternatives which I used there.

closed 2022-11-17 by edabngl

beniroquai, 2022-11-18:
@edabngl @b2jia please find the newly organized documtation for the polarbox here: https://openuc2.github.io/docs/DiscoveryKit/Polarisation/APP_POL_Brewster_Angle_Experiment/

CC: @renerichter @ranranking