openUC2/UC2-GIT #102: Circular aperture leaf too thin to print

b2jia, 2022-11-14:
Regarding `20_Cube_Insert_CirAp_Leaf_v3.stl` as requirement for the light sheet microscope, it looks like the thickness of the leaf is only 0.3mm. We have tried printing this many times to no avail. We are wondering (1) is this truly the correct thickness? (2) can it be "thicker" to work, or is the tolerance extremely small?

[image]

beniroquai, 2022-11-14:
Dear @b2jia many thanks for your message. Unfortunately, the person who developed it (@b286) no longer supports this module. Perhaps @edabngl, can you help out here? :)

b2jia, 2022-11-14:
@beniroquai thanks for your response! Just checking it is permissible to increase the thickness of this aperture leaf. I checked the clearance on the aperture wheel (onto which the hole in the leaf locks in), it's 0.6 mm in thickness. I'm thinking of both increasing the extrusion on the aperture wheel (0.6mm -> 1 mm), and the thickness of the leaf (0.3 mm -> 0.7mm).

edabngl, 2022-11-15:
Hi @b2jia , I used this design before, and I got problems with assembling the leaves to the wheel. So, my solution was the updating the wheel holder part with edges which is an up-to-date design for now. 
If you change the edge design, which you mentioned as 0.6mm -> 1 mm  (I marked this part in the image), I think you can use thicker leaves in the design. There is enough gap between the lid and guide parts for the changes. I hope it works well :)

[image]

closed 2022-11-15 by edabngl

b2jia, 2022-11-24:
For future posterity: in case others are also trying to print this aperture, 

- increasing the leaf thickness to 0.6 mm was sufficient
- increasing the clearance on the wheel (circled above) to 1 mm was also sufficient
- the clips on the wheels needed to be filed down in order for the leaves to hook on
- for future note: print the leaves with a non-transparent material (white shown below; partially translucent)

https://user-images.githubusercontent.com/37123556/203666007-abb8c5c7-1da0-4c9d-acfb-e5fa386cf78c.MOV

beniroquai, 2022-12-05:
@b2jia would you mind to file a pull request with the updated instructions? :-)

beniroquai, 2022-12-06:
And by the way this looks truely amazing! :)