openUC2/UC2-GIT #105: X-Y Stage Cube

chrisashe00, 2023-01-20:
Hi There

I'm using the same motors you are for assembling  peripheral contolled XY stage. Do you know if there is anywhere on the UC2 platform that has cube designs for this type of motor setup? 

Many Thanks.

beniroquai, 2023-01-22:
Hey @chrisashe00 thanks for reaching out. We have the CAD files available (if you'Re talking about the microsteppers). What exactly are you planning to do? PErhaps I can help.

chrisashe00, 2023-01-24:
Hey! W're planning on building a multi-fluourescence set up and using a peripheral (PS3) controller to control the XY stage. We were actually hoping to see if we could contribute to the Repo also but right now we just wanted to see where we could find a cube for mounting XY stage before we designed one ourselves.

beniroquai, 2023-01-28:
Hey @chrisashe00 sorry for late reply. We do have the STL/CAD for the xy stage. Where exactly do you wanna place the stage? Perhaps I could help you out. 
[00_Chinese_XY-table.stl.zip](https://github.com/openUC2/UC2-GIT/files/10528472/00_Chinese_XY-table.stl.zip)

[image]

chrisashe00, 2023-01-30:
Hey @beniroquai thats cool you have a cad model of the stepper itself!  so we were actually wanting to mount the stepper into a UC2 cube. Hence why we were wondering if a cube existed on the platform that could already hold the microstepper or if we'd have to make one ourselves using CAD software. Any help would be great!

beniroquai, 2023-01-30:
For me, it would be easy to do that. The question is rather how you want to
mount the sample etc.
Would you mind sketching your idea of the cube's insert eventually? This
would help me creating the insert :-)

> Message ID: ***@***.***>
>

chrisashe00, 2023-01-30:
Hi @beniroquai. Here are some sketches I drew of how the stage would sit in the cube (apologies for roughness). Unsure how we are going to mount sample in stage 
[image]
[image]

As for the sample, we have a small rectangle which will hold a slide that should sit on the stage

beniroquai, 2023-01-30:
Thanks for the sketch!
I see. From my personal experience it's be tough to focus through the small
hole of the stage. That's why we used to have two of these stages "back
then":
https://github.com/openUC2/UC2-GIT/tree/master/CAD/ASSEMBLY_XYZ-STAGE_v0

What's your sample geometry and the working distance of your lens?

> Message ID: ***@***.***>
>

chrisashe00, 2023-01-31:
Hey, again thank you for the fast response. 

We are using an infinity lens for our sample, see attached, we're unsure of exact distance but we are using this configuration, so I imagine there will be a few blocks of distance between lens and sample? (See attached skyview diagram)

Infinity Lens: https://www.amazon.ae/ZKAIAI-Professional-Biological-Microscope-Achromatic/dp/B08YXFFLRH

[image]