openUC2/UC2-GIT #106: 45 degrees lens/filter holder

stevenwlee, 2023-01-27:
Hi all,

I require a 3D-printed insert that can securely hold a circular lens like the generic lens holder but at 45 degrees (similar to the 45 degrees mirror holder). This will be specifically for circular dichroic filters.

I've attached a photo of the desired configuration of the dichroic filter cubes, I need to secure (specifically circular) dichroic mirrors at 45 degrees. Please advise me on whether a 3D-printed insert for this has been made before and if there are any STL files for this.

[image]
[image]

Thanks
Steven

beniroquai, 2023-01-28:
Hey Steven, thanks for reaching out. We do have a beamsplitter module for these filters. Perhaps you want to try that one? I attach the STL file here:
[image]

ZIP File:
[20_Cube_Insert_Beamsplittercube_Base_25x25_v3.stl.zip](https://github.com/openUC2/UC2-GIT/files/10528471/20_Cube_Insert_Beamsplittercube_Base_25x25_v3.stl.zip)

stevenwlee, 2023-01-30:
Hi Benedict,

Thanks for your reply, I forgot to mention the diameter of our lenses is 25 mm, to double check, is this suitable? 

Also, the design for the project I'm working on will require the incoming beam to be deflected to the left or right rather than upwards. Thanks very much for sending me the above STL file. I will need some of this insert to be turned 90 degrees so that the filter can be angled similarly to my picture. I need to take the model in your picture and rotate it 90 degrees sideways before changing the sides so it will still lock into a 3dp cube. Does this configuration already exist? 

If not might need to edit this using CAD. And by doing that I might also need a way to tighten the lens on this so it doesn't fall, a bit like the generic lens holder with its removable cover.

Best Regards
Steven

beniroquai, 2023-01-30:
Hey @stevenwlee thanks for your message. I guess I do not fully understand the requirements by your design - sorry ;-) 
Would you mind sketching your idea on paper? I can try designing it in 3D. 
Also, what's the thickness of the lens? (sorry though it's a filter..) Should the lens's center be in the middle of the cube?

stevenwlee, 2023-01-31:
Hi Benedict, thank you very much for your message, The dichroic mirrors I am using all have a thickness of 1 mm (and all have a 25 mm diameter). Please see the drawing of the cube design(shaded) i want to model, and to be clear this it is a sky view perspective.
[image]

I aim to make the incoming light deflected by 90 degrees, but not upwards, but instead right or left. The dashed lines represent where i'd make a hole for line to enter or exit the cube. The dichroic mirror will be angled at 45 degrees, I hope this picture makes it clearer to you. The diagram that my teammate @chrisashe00 also attached in a separate issue also demonstrates how I want to position my dichroic filters (part of our multifluorescence setup). 

[image]

Is there an ipt file for the beam splitter cube you originally sent?