openUC2/UC2-GIT #111: Thorslab Tube Lens and Cage System Specifications?

jsmithe2, 2024-08-22:
Hello @beniroquai 

https://github.com/openUC2/UC2-GIT/tree/dea2aca9f7fb344b781abce6a466f22229fd3c4d/CAD/ASSEMBLY_CUBE_Thorlabs

I am trying to make this thorslab cage cube for my Allied Vision Alvium 1800 U-158m. From the information I have here, the cage system is Thorslab CP36 but what would be the proper tube lens for this attachment into the UC2 cube system for this camera?

beniroquai, 2024-08-22:
We use the "ordinary" cage system (e.g. https://www.thorlabs.com/newgrouppage9.cfm?objectgroup_id=4114). This would fit into the cube using this insert. 
As for the tube lens I would rather recommend you to use either a CCTV lens or o 150mm lens depending on the NA you would use and the corresponding sampling criteria on this camera.

jsmithe2, 2024-08-22:
Are you able to provide a link or product number to the types of lenses you are talking about? I understand I will have to adjust based on the NA I decide on.

jsmithe2, 2024-08-23:
@beniroquai More specifically, which tube lens did you use in this picture? Is this a thorslab tube lens?

[image]

beniroquai, 2024-08-30:
Ah, sorry for my late reply. Exactly. This is an achromat from Thorlabs. It could be any, but we used 50,100, 150mm etc. since this is compatible with our standard :)