openUC2/UC2-GIT #112: Baseplate

Grmpfff, 2025-01-09:
Dear UC2 Team,

I would like to print the bigger baseplates that are shown on the pictures to ensure the better stability for towerlike setups (for example the smartphone mikroskope). I regulary saw pictures of the 4x2 and 4x4 baseplate but couldnt find them in the RAW folder or the configurator.
Can anyone provide the stl?
Kind Regards
Benjamin

beniroquai, 2025-01-16:
Hey @Grmpfff sorry for my late reply. A very old version can be found here
https://github.com/openUC2/UC2-GIT/blob/133475f21ddba18bf52d852c6b364a27358937ea/CAD/RAW/STL/Assembly_ALL_PARTS_FOR_EXPORT_10_Base_v0_1.stl
You can arrange them in the slicer software with 50mm spacing so then they will become 1 big baseplate (same for the puzzle pieces by the way). Was this what you needed? 

[image]

Best
Bene

Grmpfff, 2025-01-19:
Thanks very much! I didn't find this folder. I will start looking for the mold cube model there that doesn't need any screws.

beniroquai, 2025-01-19:
Cool! Let us know if you need anything over at https://openuc2.discourse.group/. We hope to create the community there, now :-)

closed 2025-01-19 by beniroquai